asyncvlsi/act #1: Initial configuration does not seem to work

dmzimmerman, 2019-03-21:
There appears to be a circular dependency, whereby the `configure` script fails to create a `config` script in the `scripts` directory because there is no `config` script in the `scripts` directory.

More specifically, the `check` script called by the `configure` script fails with:
```
line 16: /Users/dmz/misc-repos/act/scripts/config: No such file or directory
```

dmzimmerman, 2019-03-21:
As a side note, putting a stub `config` script in the `scripts` directory that defines the `C_COMPILER` environment variable properly allows configuration (and building) to proceed.

rmanohar, 2019-03-21:
Thanks! I believe this has now been fixed.

dmzimmerman, 2019-03-21:
I can confirm that it now works on a freshly checked out repo. Thanks!

closed 2019-03-21 by dmzimmerman