asyncvlsi/act #5: Configuration file location and options need to be cleaned up and documented

nbingham1, 2020-08-03:
```
/opt/cad$ echo $ACT_HOME
/opt/cad
/opt/cad$ ls -l
total 20
drwxr-xr-x 15 4096 Aug  2 13:19 act
drwxr-xr-x  2 4096 Aug  3 11:37 bin
drwxr-xr-x  3 4096 Aug  3 11:42 conf
drwxr-xr-x  3 4096 Aug  2 22:05 include
drwxr-xr-x  2 4096 Aug  3 10:48 lib
/opt/cad$ cd conf/
/opt/cad/conf$ ls -l
total 12
drwxr-xr-x 2 4096 Aug  3 11:39 generic
drwxr-xr-x 2 4096 Aug  3 13:23 st28
-rw-r--r-- 1 2086 Aug  3 13:26 st28.conf
/opt/cad/conf$ cd st28/
/opt/cad/conf/st28$ ls -l
total 16
-rw-r--r-- 1   59 Aug  3 13:23 extract.conf
-rw-r--r-- 1 3292 Aug  3 13:23 global.conf
-rw-r--r-- 1   59 Aug  3 13:23 lint.conf
-rw-r--r-- 1 2086 Aug  3 11:38 prs2net.conf
```

None of these work
```
/opt/cad/conf$ pushd
~/act/test/prs/clk_mult_dadda___64_64_6______bits /opt/cad/conf
~/act/test/prs/clk_mult_dadda___64_64_6______bits$ prs2net -C st28 -p'testbench<>' test.act 
FATAL: Could not open configuration file `st28.conf' for reading.
~/act/test/prs/clk_mult_dadda___64_64_6______bits$ prs2net -T st28 -p'testbench<>' test.act 
FATAL: Could not open configuration file `global.conf' for reading.
~/act/test/prs/clk_mult_dadda___64_64_6______bits$ prs2net -T /opt/cad/conf/st28 -p'testbench<>' test.act 
FATAL: Could not open configuration file `global.conf' for reading.
nbingham@Lira:~/act/test/prs/clk_mult_dadda___64_64_6______bits$ prs2net -p "testbench<>" -Tst28 test.act
prs2net: invalid option -- 'T'
Unknown option.
Usage: prs2net [-dltBR] [-C <conf>] [-p <proc>] [-o <file>] <act>
 -C <conf> Configuration file name
 -c <cells> Cell file name
 -t        Only emit top-level cell (no sub-cells)
 -p <proc> Emit process <proc>
 -o <file> Save result to <file> rather than stdout
 -d	       Emit parasitic source/drain diffusion area/perimeters with fets
 -B	       Turn of black-box mode. Assume empty act process is an externally specified file
 -l	       LVS netlist; ignore all load capacitances
 -S        Enable shared long-channel devices in staticizers
```

This is the one that works.
```
prs2net -Tst28 -p "testbench<>" test.act
```

rmanohar, 2020-08-03:
Documented here now.  https://avlsi.csl.yale.edu/act/doku.php?id=config:start

rmanohar, 2020-08-03:
Also modified usage information for some of the core act tools.

closed 2020-08-03 by rmanohar