asyncvlsi/act #8: Old loop styles don't generate parser errors

nbingham1, 2020-08-03:
```
defproc test(bool a[10], b[10], c, d)
{
  (;i:10:
    prs {
      a[i] => b[i]-
    }
  )

  prs {
    (|:i:10: a[i] ) => c-
  }
}

test t;
```

```
$ aflat test.act
"t.a[0]"->"t.b[0]"-
~("t.a[0]")->"t.b[0]"+
"t.a[1]"->"t.b[1]"-
~("t.a[1]")->"t.b[1]"+
"t.a[2]"->"t.b[2]"-
~("t.a[2]")->"t.b[2]"+
"t.a[3]"->"t.b[3]"-
~("t.a[3]")->"t.b[3]"+
"t.a[4]"->"t.b[4]"-
~("t.a[4]")->"t.b[4]"+
"t.a[5]"->"t.b[5]"-
~("t.a[5]")->"t.b[5]"+
"t.a[6]"->"t.b[6]"-
~("t.a[6]")->"t.b[6]"+
"t.a[7]"->"t.b[7]"-
~("t.a[7]")->"t.b[7]"+
"t.a[8]"->"t.b[8]"-
~("t.a[8]")->"t.b[8]"+
"t.a[9]"->"t.b[9]"-
~("t.a[9]")->"t.b[9]"+
"t.a[0]"|"t.a[1]"|"t.a[2]"|"t.a[3]"|"t.a[4]"|"t.a[5]"|"t.a[6]"|"t.a[7]"|"t.a[8]"|"t.a[9]"->"t.c"-
~("t.a[0]"|"t.a[1]"|"t.a[2]"|"t.a[3]"|"t.a[4]"|"t.a[5]"|"t.a[6]"|"t.a[7]"|"t.a[8]"|"t.a[9]")->"t.c"+
```

rmanohar, 2020-08-03:
The `(;i:...)` construct is still currently supported, although it is deprecated as it's not really the correct syntactic replication construction. The `(|i:...)` construct is right; what doesn't work is the older syntax of `(:|i:...)`

nbingham1, 2020-08-03:
The syntax I have written is `(|:i:...)` with the extra colon just before the variable name. Perhaps a warning about future deprecation would help. Low priority anyway, just figured that I'd make note of it. :)

rmanohar, 2020-08-03:
It now complains about a parse error for the `(|:i:...)`

closed 2020-08-03 by nbingham1