asyncvlsi/act #10: aflat ignores exclhi and excllo

nbingham1, 2020-08-10:
```
defproc test(bool a, b)
{
	spec {
		exclhi(a, b)
	}
	prs
	{
		a => b-
	}
}

test t;
```

```
$ aflat tmp.act 
"t.a"->"t.b"-
~("t.a")->"t.b"+
```

rmanohar, 2020-08-10:
That's correct behavior. You're supposed to say  mk_exclhi(a,b). The "exclhi(a,b)" is viewed as something that the circuit has to guarantee on its own (e.g. for a dualrail code).

nbingham1, 2020-08-10:
Isn't prsim supposed to issue a warning if the "exclhi" spec is violated?

rmanohar, 2020-08-10:
No, they are simply stripped out for prsim, as it doesn't have any support for invariant assertions.

nbingham1, 2020-08-10:
So what are exclhi and excllo connected to then? Is it just a TODO at the moment?

rmanohar, 2020-08-10:
They are used by lvp. More generally, the entire spec section will be re-done in the next year.

closed 2020-08-10 by nbingham1