asyncvlsi/act #11: Technology config file doesn't support absolute path

nbingham1, 2020-08-10:
Someone may need to edit the config files or use their own. However, the group directory isn't writeable (for good reason). It might be a good idea for the -T option to support absolute paths so that people can point the tools to their own config files.

```
defproc test(bool a, b)
{
  prs {
    a => b-
  }
}

test t;
```

```
$ pwd
/home/eb823/cad/conf/generic
$ ls -l
total 2
-rw-r--r-- 1   59 Aug 10 15:52 extract.conf
-rw-r--r-- 1 3292 Aug 10 15:52 global.conf
-rw-r--r-- 1   59 Aug 10 15:52 lint.conf
-rw-r--r-- 1   59 Aug 10 15:52 prs2net.conf
$ echo $ACT_HOME
/home/eb823/cad
```

```
$ prs2net -T/home/eb823/cad/conf/generic -p'test<>' tmp.act
FATAL: Could not open configuration file `global.conf' for reading.
```

rmanohar, 2020-08-10:
If you've modified ACT_HOME, then presumably this would work without -T?

nbingham1, 2020-08-10:
Don't other things depend on ACT_HOME? or is it just the config files?

rmanohar, 2020-08-10:
ACT_HOME is used to access ACT files and config files. The ACT files can be also accessed using ACT_PATH.

rmanohar, 2021-10-19:
This should work now.