asyncvlsi/act #12: prs2net Signal connections to Vdd or GND don't work

nbingham1, 2020-08-11:
This seems like it may be an intentional thing, mandating a tie-down cell? If so, prs2net should emit a warning.
```
b.c.d[0] = g.Vdd;
```

```
Warning: node "xdut.xdut.sub_ab_ac_ad_50_6" has no DC path to ground
```

rmanohar, 2020-08-11:
This looks like a warning being reported by hsim/HSPICE because of the generated circuit.
ACT is not responsible for warning you about the limitations of third-party tools.

nbingham1, 2020-08-12:
This was not a problem in the earlier version of act. This warning in spice comes from a change in the way the spice file is generated. IE connecting a signal to vdd or ground is not implemented in the new version of act, but is in the old version

rmanohar, 2020-08-12:
Thre is nothing special about "g.GND". It is just a name. Please provide a concrete example that shows that prs2net is producing an incorrect output.

nbingham1, 2020-08-12:
I'll email you the whole directory, but for future reference:

### wchb.act
```
export deftype globals <: int<6> (bool Vdd, GND, vpsub, vnsub, Reset, _Reset, pReset, sReset, _pReset, _sReset)
{
	sReset = Reset;
	_sReset = _Reset;
}

// PN ratio
//preal PN = 1.4142;
preal PN = 2.0;

// Transistor minimal sizing
pint NM = 6;
pint PM = 6;

export template <pint N>
deftype c1of <: int (bool d[N])
{
}

export template <pint N>
defchan e1of <: chan(int) (c1of<N> b; bool d[N]; bool e)
{
	d = b.d;
}

export defproc wchb1(globals g; e1of<2> L, R)
{
	c1of<1> _R;
	R.d[1] = g.GND;
	prs<g.Vdd, g.GND>
	{
		g._sReset & R.e & L.d[0] -> _R.d[0]-
		~_R.d[0] -> R.d[0]+

		R.d[0] | L.d[1] -> L.e-

		~g._pReset | ~R.e & ~L.d[0] -> _R.d[0]+
		_R.d[0] -> R.d[0]-
		
		~R.d[0] & ~L.d[1] -> L.e+
	}
}

export defproc wchb2(globals g; e1of<2> L, R)
{
	c1of<2> _R;
	prs<g.Vdd, g.GND>
	{
		g._sReset & R.e & L.d[0] -> _R.d[0]-
		g._sReset & R.e & L.d[1] -> _R.d[1]-
		~_R.d[0] -> R.d[0]+
		~_R.d[1] -> R.d[1]+

		R.d[0] | R.d[1] -> L.e-

		~g._pReset | ~R.e & ~L.d[0] -> _R.d[0]+
		~g._pReset | ~R.e & ~L.d[1] -> _R.d[1]+
		_R.d[0] -> R.d[0]-
		_R.d[1] -> R.d[1]-
		
		~R.d[0] & ~R.d[1] -> L.e+
	}
}

export defproc fifo(globals g; e1of<2> L, R)
{
[... 172 lines cut ...]
*.PININFO g_aVdd:I g_aGND:I g_a__Reset:I g_a__pReset:I L_ab_ad_50_6:I L_ab_ad_51_6:I L_ae:O R_ab_ad_50_6:O R_ab_ad_51_6:O R_ae:I
*.POWER VDD Vdd
*.POWER GND GND
*.POWER NSUB GND
*.POWER PSUB Vdd
xbuf1 g_aVdd g_aGND g_a__Reset g_a__pReset M_ad_50_6 **M_ad_51_6** M_ae R_ab_ad_50_6 R_ab_ad_51_6 R_ae wchb2
xbuf0 g_aVdd g_aGND g_a__Reset g_a__pReset L_ab_ad_50_6 L_ab_ad_51_6 L_ae M_ad_50_6 M_ae wchb1
.ends
*---- end of process: fifo<> -----

.subckt wrapper g_a__Reset g_a__pReset L_ab_ad_50_6 L_ab_ad_51_6 L_ae R_ab_ad_50_6 R_ab_ad_51_6 R_ae
xdut g_aVdd g_aGND g_a__Reset g_a__pReset L_ab_ad_50_6 L_ab_ad_51_6 L_ae R_ab_ad_50_6 R_ab_ad_51_6 R_ae fifo
.ends
```

```
Warning: node "xdut.xdut.m_ad_51_6" has no DC path to ground
Warning: 1 node with connectivity problems are reported in 'hsim.conn'
```

```
*Nodes with no DC path to ground or less than 2 connections
xdut.xdut.m_ad_51_6 - no DC path
 wrapper fifo
```

rmanohar, 2020-08-12:
This has been closed for not conforming to the [posted guidelines](https://github.com/asyncvlsi/act/blob/master/CONTRIBUTING.md) multiple times.

rmanohar, 2020-08-12:
From a cursory  glance at this,  there seems to be a problem with
```
.subckt wrapper g_a__Reset g_a__pReset L_ab_ad_50_6 L_ab_ad_51_6 L_ae R_ab_ad_50_6 R_ab_ad_51_6 R_ae
xdut g_aVdd g_aGND g_a__Reset g_a__pReset L_ab_ad_50_6 L_ab_ad_51_6 L_ae R_ab_ad_50_6 R_ab_ad_51_6 R_ae fifo
.ends
```
The Vdd and GND signals are not being exported to the `wrapper` port list? The SPICE netlist above looks like it was generated by `prs2net`; not sure what generated this particular definition of `wrapper`.

nbingham1, 2020-08-12:
`g_aVdd` and `g_aGND` are globals that are connected to DC power sources in `test.spi` as generated by prspice.

```
.global g_aVdd g_aGND
vpwr0 g_aVdd 0 dc 1.0v
vpwr1 g_aGND 0 dc 0.0v
```

The bug lies in the `fifo` process definition in `dut.spi`.

```
xbuf1 g_aVdd g_aGND g_a__Reset g_a__pReset M_ad_50_6 M_ad_51_6 M_ae R_ab_ad_50_6 R_ab_ad_51_6 R_ae wchb2
xbuf0 g_aVdd g_aGND g_a__Reset g_a__pReset L_ab_ad_50_6 L_ab_ad_51_6 L_ae M_ad_50_6 M_ae wchb1
```

specifically, `M_ad_51_6` should be replaced with `g_aGND`  when instantiating `xbuf1` because it was `M_ad_51_6` was eliminated from `xbuf0` and is therefore no longer connected to anything.

rmanohar, 2020-08-12:
Ok I found a comment in the act source that says "fix this" which I think corresponds to the issue. Also created a new

rmanohar, 2020-08-12:
... also created a new, much smaller test case.

rmanohar, 2020-08-13:
Just pushed a fix.

closed 2020-08-13 by nbingham1