asyncvlsi/act #15: Canonical Name Issue

nbingham1, 2020-09-21:
**Describe the bug**
This is another canonical name bug. prs2net is choosing different names for the same signal in to places of the spice output. This causes them to be disconnected.

**To Reproduce**
```
prs2net -p'myproc3<>' test.act
```

**ACT source (test.act)**
```
defchan mytype <: chan(int) (bool d)
{
}

defchan mytype2 <: chan(int) (mytype c)
{
}

defproc myproc1(bool a, b)
{
	prs
	{
		a => b-
	}
}

defproc myproc2(bool a; mytype2 A; mytype B)
{
	A.c.d = B.d;

	prs
	{
		a => B.d-
	}
}

defproc myproc3(mytype A)
{
	mytype2 B;
	mytype C;

	B.c = A;

	myproc2 x(, B, C);
	myproc1 y(C.d);
}

myproc3 tst;
```

**Expected behavior**
On line 53 and 54:
```
xy C_ad y_ab myproc1
xx x_aa A_ad myproc2
```
`C_ad` and `A_ad` should be connected.

**Computer setup (please complete the following information):**
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-39)

rmanohar, 2020-09-23:
Nice short test case. I pushed a fix for this issue.

nbingham1, 2020-09-23:
Fix works :)

closed 2020-09-23 by nbingham1