asyncvlsi/act #16: The "A first ACT tutorial" link is broken

tommythorn, 2020-12-01:
**Describe the bug**
Following the ACT tutorial link (http://avlsi.csl.yale.edu/act/doku.php?id=tutorial:basicprs) leads to a page proclaiming
```
DokuWiki Setup Error

The datadir ('pages') at /home/avlsi_wiki/data_act/pages is not found, isn't accessible or writable. You should check your config and permission settings. Or maybe you want to run the installer?
```

**To Reproduce**
above

**ACT source**
N/A

**Expected behavior**
Obvious

**Screenshots**
above

**Computer setup (please complete the following information):**
Firefox & Safari both agree

**Additional context**
N/A

rmanohar, 2020-12-01:
Thanks, we know. The machine hosting this server had a disk failure and people are working on recovering the files... hope to get this fixed shortly.

rmanohar, 2020-12-04:
Just an update. The server is back, the links should work again. We will be migrating this over to EC2 shortly, and will update links once that migration is complete.

tommythorn, 2020-12-04:
Awesome, thanks.  Looking forward to your talk Tuesday.

closed 2020-12-04 by tommythorn