asyncvlsi/act #18: Transistor length doesn't support pint expressions

nbingham1, 2020-12-24:
**Describe the bug**
When using a pint expression to set the transistor length, aflat produces a parse error about transistor flavor. Notice that replacing "B" with 20 does not produce this error.

**To Reproduce**
Steps to reproduce the behavior:
1.  Use the ACT source included below.
2. Run the following sequence of commands: `aflat test.act`
3. Observe that the following parse error occurs.
```
Parse error: Could not parse entire file.
Error parsing production rule: unknown transistor flavor
	File `test.act', line: 8, col: 9.
Exception: number=1, message -none-
FATAL: Uncaught exception
```

**ACT source**
test.act:
```
pint A = 6;
pint B = 20;

defproc test()
{
  bool a, b;
  prs {
    a<A,B> -> b-
    ~a<A,B> -> b+
  }
}

test t;
```

**Expected behavior**
aflat succeeds and emits the flattened prs.

**Computer setup (please complete the following information):**
Ubuntu 20.10
gcc 10.2.0
GNU Make 4.3
ACT commit a75aa488c29edb8e8b29fa1e215c54793b9582cc

rmanohar, 2020-12-24:
Should be fixed now.

nbingham1, 2020-12-25:
Yep, its fixed. Thanks!

closed 2020-12-25 by nbingham1