asyncvlsi/act #21: prsim: promt does not appear

olerichter, 2021-04-06:
**Describe the bug**
when starting prsim without anything the promt does not appear
when running with act file

**To Reproduce**
Steps to reproduce the behavior:
compile fresh of the git both mac or centos
run prsim
get an unresponsive terminal
or 
run prsim -r "test.act"

```
defproc a0() {

  chp {
	skip
  }
}
```
get 
FATAL: lex_mustbe: expected token: ->, received: -identifier-
	File: test.act, line: 1, col: 11

**Expected behavior**
work as before :)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Computer setup (please complete the following information):**
 - centos7 with gcc7.3
 - macos 10.15 with clang 12.0.0
 - grace

rmanohar, 2021-04-06:
Strange. Can you try it with a .prs file? I seem to get a prompt... I used a simple file
```
a -> b-
```
and I see the same expected behavior.

olerichter, 2021-04-06:
sorry may brain short circuited a bit, i was trying to simplify+reproduce a different issue, and did not realise that it obviously does not make sense to run prsim with act files..., somewhere on my reproduction tries i must have switched.

but yes prsim -r test.prs (so not act content) works fine

olerichter, 2021-04-06:
I know now what trew me of,
if you ctrl-C on prsim, the terminal becomes unresponsive but does not terminate,
i will close this issue

closed 2021-04-06 by olerichter