asyncvlsi/act #23: netlisting has trouble with positive labels for distributed shared gate network

nbingham1, 2021-05-22:
**Describe the bug**
unexpected error message for shared gate network

**To Reproduce**
Steps to reproduce the behavior:
1.  Use the ACT source included below.
2. `prs2net -p'test<>' temp.act`
3. produces an error message `FATAL: @-expression with unknown label 'x'`

**ACT source**
temp.act
```
defproc test() {
  bool a, b, c;
  prs {
    ~a -> @x+
  }

  (;i:2:
    prs {
      @x & ~b -> c+
    }
  )
}

test t;
```

**Expected behavior**
expected netlist output

**Computer setup (please complete the following information):**
g++ (Ubuntu 10.2.0-13ubuntu1) 10.2.0
LSB Version:	core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 20.10
Release:	20.10
Codename:	groovy

**Additional context**
inverting the sense of the label works properly

rmanohar, 2021-05-24:
Thanks. I pushed a fix.

nbingham1, 2021-05-24:
Yep, it works

closed 2021-05-24 by nbingham1