asyncvlsi/act #24: binary constants in v2act

olerichter, 2021-05-25:
**Describe the bug**

a) v2act_quote should be for handling more than 9 bits :) (i added a + after the first number bracket)  **proposed fix**``` s/\([0-9]+\)'b\([01][01][01]*\)/\1'b"\2"/g ```

b) because v2act_quote.sed is marked as a target in the makefile it is deleted if you run make realclean 
so it needs to be restored from git, **Wanted behaviour**: dont delete this target

c) the binary numbers are read LSB to MSB should be MSB to LSB:
```
module blk0(out);
  output [7:0] out;
  assign out = 8'b00111100;
endmodule
```
**results in**
```
export defproc blk0 (syn::sdtexprchan<1> out[8])
{
   /*--- types ---*/
   source_GND src_GND_0;
   source_GND src_GND_1;
   source_GND src_GND_2;
   source_GND src_GND_3;
   source_Vdd src_Vdd_0;
   source_Vdd src_Vdd_1;
   source_Vdd src_Vdd_2;
   source_Vdd src_Vdd_3;
   /*--- connections ---*/
   out[7]=src_GND_0.out;
   out[6]=src_GND_1.out;
   out[5]=src_Vdd_0.out;
   out[4]=src_Vdd_1.out;
   out[3]=src_Vdd_2.out;
   out[2]=src_Vdd_3.out;
   out[1]=src_GND_2.out;
   out[0]=src_GND_3.out;
}
```
but should be **wanted behaviour**
```
export defproc blk0 (syn::sdtexprchan<1> out[8])
{
   /*--- types ---*/
   source_GND src_GND_0;
   source_GND src_GND_1;
   source_GND src_GND_2;
   source_GND src_GND_3;
   source_Vdd src_Vdd_0;
   source_Vdd src_Vdd_1;
   source_Vdd src_Vdd_2;
   source_Vdd src_Vdd_3;
   /*--- connections ---*/
   out[7]=src_GND_3.out;
   out[6]=src_GND_2.out;
   out[5]=src_Vdd_3.out;
   out[4]=src_Vdd_2.out;
   out[3]=src_Vdd_1.out;
   out[2]=src_Vdd_0.out;
   out[1]=src_GND_1.out;
   out[0]=src_GND_0.out;
}
```

rmanohar, 2021-05-25:
Fixed. (sed script is fine, but the bits were reversed.)

olerichter, 2021-05-26:
works!

closed 2021-05-26 by olerichter