asyncvlsi/act #25: support for 64bit ints in miniscm

olerichter, 2021-05-28:
**Describe the bug**
when running prsim, and I use integers with more 32bit while invocing the scheme interpreter it will only use the first 32bits

**To Reproduce**

take the 3 files attached and run `prsim 64.prs` and then `source 64.prsim` while walking past 32 bits it will give:
```
Value must be `0', `1', or `X'
Execution aborted.
Stack trace:
	called from: assert
	called from: begin
	called from: let
	called from: assert-var-bit
	called from: begin
	called from: cond
	called from: assert-one-bit
	called from: letrec
	called from: assert-var-int
	called from: source
```
[Archive.zip](https://github.com/asyncvlsi/act/files/6563341/Archive.zip)

**Expected behavior**
same behavior as for numbers >32bit

like `assert-var-int "t.x" 32 75808`

**Additional context**
I tired fixing it myself, but yea i am not there yet haha

olerichter, 2021-05-28:
reason: so i can write a 64bit test for expropt

rmanohar, 2021-05-29:
Odd, I cannot reproduce the error you're getting on a Linux box (RHEL) or MacOS. But miniscm is using 32-bit integers internally. Will make the modification but you will have to let me know if the fix works.

closed 2021-09-22 by rmanohar