asyncvlsi/act #27: pfet shows error for violating nfet max width, perhaps integer overflow as well

nbingham1, 2021-09-22:
**Describe the bug**
Act incorrectly errors against the maximum nfet width limit when writing the netlist for a pfet transistor for the skywater 130 open source pdk.

Here are the relevent global.conf rules (whole file posted below)
```
int  min_width  12
int  min_length 5

int max_n_width 3333
int max_p_width 233
```

```
$ prs2net -Tsky130 -DDUT=true -DLAYOUT=true -DPRSIM=false  -p "wchb<>" wchb.act
...
In expanding wchb<> (wchb.act:17)
FATAL: Device #13: pfet width (12) exceeds maximum limit (3333)
```

**To Reproduce**
Steps to reproduce the behavior:
1.  Use the ACT source included below.
2. Run the following sequence of commands
```
prs2net -Tsky130 -DDUT=true -DLAYOUT=true -DPRSIM=false  -p "wchb<>" wchb.act
```
3. Observe that the above error occurs.

**ACT source**
global.conf
```
begin act
string_table dev_flavors "svt" "lvt" "hvt" "mvt"
int max_recurse_depth 1000
string_table spec_types "exclhi" "excllo" "mk_exclhi" "mk_excllo"
string_table prs_attr "i:s:after" "i:s:keeper" "i:s:iskeeper" "i:s:isckeeper" "r:+:loadcap" "i:M:oresis" "i:s:output" "r:s:N_reff" "r:s:P_reff" "i:s:weak" "i:s:unstab" "i:s:comb"
string_table instance_attr "i:s:pos_x" "i:s:pos_y" "i:s:size_x" "i:s:size_y" "i:s:autokeeper" "i:s:iscomb"
string mangle_chars ":()<>[],{}.""
string mangle_letter "_"
end
begin net
string name "scmos"
string_table ext_devs "nfet" "pfet"
string_table ext_map  "nfet_svt" "pfet_svt"
string spice_path_sep ":"
int  std_p_width  14
int  std_p_length 5
int  std_n_width  12
int  std_n_length 5
int  min_width  12
int  min_length 5
int max_n_width 3333
int max_p_width 233
int  stat_p_width 14
int  stat_p_length 5
int  stat_n_width 12
int  stat_n_length 5
int fet_spacing_diffonly 12
int fet_spacing_diffcontact 12
int fet_diff_overhang 10
real p_n_ratio 2.0
real weak_to_strong_ratio 0.1
real lambda 0.03
real default_load_cap 1.0
string extra_fet_string ""
int disable_keepers 0
int discrete_length 0
int swap_source_drain 0
int use_subckt_models 1
int fold_pfet_width 0
int fold_nfet_width 0
int ignore_loadcap 1
int emit_parasitics 0
string pfet_svt "sky130_fd_pr__pfet_01v8"
string pfet_lvt "sky130_fd_pr__pfet_01v8_lvt"
string pfet_hvt "sky130_fd_pr__pfet_01v8_hvt"
string pfet_hvt "sky130_fd_pr__pfet_01v8_mvt"
string nfet_svt "sky130_fd_pr__nfet_01v8"
string nfet_lvt "sky130_fd_pr__nfet_01v8_lvt"
string nfet_hvt "sky130_fd_pr__nfet_01v8_hvt"
string nfet_hvt "sky130_fd_pr__nfet_01v8_mvt"
string global_vdd "Vdd"
string global_gnd "GND"
string local_vdd "VddN"
string local_gnd "GNDN"
begin sizing
  int unit_n 5
  int p_n_mode 0
end
end  
begin lint
real Vdd 1.8
int verbose 0
real slewrate_fast_threshold 200.0
real slewrate_slow_threshold 20.0
real V_high 1.5
real V_low  0.3
int max_print 100
int show_frequency 1
string filter_results ""
real hysteresis 0.0
real skip_initial_time 0.0
real reset_sync_time 20.0e-9
end
begin lvp
real Vtn 0.3
real Vtp 1.5
real WeakByWidth 3.1
real NResetPStat 5.0
real PResetNStat 20.0
real CombThreshold  2.0
real StateThreshold 0.8
end
```

wchb.act
```
export deftype globals <: int<6> (bool Vdd, GND, vpsub, vnsub, Reset, _Reset, pReset, sReset, _pReset, _sReset)
{
  sReset = Reset;
  _sReset = _Reset;
}

export template <pint N>
deftype c1of <: int (bool d[N])
{
}

export template <pint N>
defchan e1of <: chan(int) (bool d[N]; bool e)
{
}

export defproc wchb(globals g; e1of<2> L, R)
{
  c1of<2> _R;
  bool Rv;
  prs<g.Vdd, g.GND>
  {
    g._sReset & R.e & L.e & L.d[0] -> _R.d[0]-
    g._sReset & R.e & L.e & L.d[1] -> _R.d[1]-
    g._sReset & Rv -> L.e-
    ~g._pReset | ~R.e & ~L.e & ~L.d[0] -> _R.d[0]+
    ~g._pReset | ~R.e & ~L.e & ~L.d[1] -> _R.d[1]+
    ~g._pReset | ~Rv -> L.e+
    ~_R.d[0] -> R.d[0]+
    _R.d[0] -> R.d[0]-
    ~_R.d[1] -> R.d[1]+
    _R.d[1] -> R.d[1]-
    ~_R.d[0]|~_R.d[1] -> Rv+
    _R.d[0]&_R.d[1] -> Rv-
  }
}

globals g;
e1of<2> L, R;
wchb dut(g, L, R);
```

**Expected behavior**
prs2net should complete without errors

**Computer setup (please complete the following information):**
```
$ lsb_release -a
LSB Version:	core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 21.04
Release:	21.04
Codename:	hirsute
$ g++ --version
g++ (Ubuntu 10.3.0-1ubuntu1) 10.3.0
```

rmanohar, 2021-09-22:
Two changes have been made and pushed.

(1) There was an error message in the fet type (both n/p fet errors were printed out as pfet; now fixed). 

(2) The second issue is that width and length parameters are maintained as an integer multiple of the manufacturing grid (default is 5nm). The configuration file settings have lambda=0.03 meters used to work-around the issue that Skywater's model deck includes a global scale factor of 1e-6, forcing all input spice decks to multiply device w/l by 1e6 to compensate.

Instead, use what you actually mean: lambda=0.03e-6. That will get rid of the error; the error was caused by an integer overflow in width/length calculations. A new option "-s <scale>" has been added to prs2net to scale all transistor parameters by the specified scale factor. So you can workaround the scale factor in the Skywater technology setup using "-s 1e6" as an option to prs2net.

nbingham1, 2021-09-22:
It works, that was amazingly fast. Reported at 10pm, fixed by 6am o.O

Is it possible to associate the scale factor directly with the configuration file instead of having it as a CLI option?

rmanohar, 2021-09-22:
Yup, you can do that right now. Use:

`begin net
real output_scale_factor 1e6
end
`

nbingham1, 2021-09-22:
Perfect, thank you!

closed 2021-09-22 by nbingham1