asyncvlsi/act #34: At least one node required for subcircuit test

nbingham1, 2021-11-28:
**Describe the bug**
prs2net creates stubs in the spice file for empty processes. Xyce does not allow empty processes. The following is the simplest example, but this most often shows up in templated processes like delay lines. For example, a delay line of length 0.

**To Reproduce**
Steps to reproduce the behavior:
1.  Use the ACT source included below.
2. `prs2net -Tsky130 -p 'test<>' test.act > test.spi`
3. Observe the following spice file
```
*
*---- act defproc: test<> -----
* raw ports: 
*
.subckt test
*.POWER VDD Vdd
*.POWER GND GND
*.POWER NSUB GND
*.POWER PSUB Vdd
.ends
*---- end of process: test<> -----
```
4. `Xyce test.spi`
5. Observe
```
Netlist error in file test.spi at or near line 5
At least one node required for subcircuit test
```

**ACT source**
```
export defproc test()
{
	prs {
	}
}

test dut;
```

**Expected behavior**
prs2net should create an empty spice file

**Computer setup (please complete the following information):**
```
$ lsb_release -a; g++ --version
LSB Version:	core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 21.04
Release:	21.04
Codename:	hirsute
g++ (Ubuntu 10.3.0-1ubuntu1) 10.3.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

rmanohar, 2021-11-30:
Just pushed a fix for this.

nbingham1, 2021-12-02:
I can confirm it works, thanks!

closed 2021-12-02 by nbingham1