asyncvlsi/act #42: [act2v] add flag to enable compatiblilty mode with cadence xelcium/spectre netlister and simulatior plus innovus/virtuoso

olerichter, 2022-04-27:
@rmanohar I will implement it but i wanted to get your ok/opinion first. also any wishes for the char that i use as flag?

 **Describe the bug**
cadence xelium/spectre and for some parts innovus does not understand or creates problems using the verilog from act2v:

1. the signal names can not contain dots so eg. 'buffer.in.a' it confuses the tool as it interprets it as a hierarchy and does not find the instances/modules, the escape slash in front of the name is ignored :/ .
2. the some parts of virtuoso need '[ ]' mangled/removed/changed, innovus is fine it ignores the brackets and just treat them as a name, so a name 'buffer.in.d[1].d[0]' would create problems as to tool starts escaping them inconsistently and therefor cuts connections.
3. the AMS xelcium simulator does not understand multiline signal specifications so
```
output signal1;
reg signal1;
```
needs to be 
```
output reg signal1;
```
4. is minor global signals (we dont use them) are defined as globals.<signalname> it does not understand top.<signalname> 

**options for later**
2. i thought to make it readable and more user friendly in virtuoso (if every single bus signal is separate we have top level module definitions with 10k signals) to do a conversion like this:
as virtuoso really only understands one dimensional arrays:

if its a uniform nested array signal like `blub.d[].d[].blub` resort them into a one dimensional with the and change to name to `blub.d[dimA].d[dimB].blub [dimA*dimB]` and do the mapping simply lowest level as least significant

```
# if we have a aMx1of2<32> called in t have (X = .,Y = [,Z = ] is a place holders for mangling)
module buffer (in_Xa, in_Xd_Y32_Z_Xd_Y2_Z)
input in_Xa;
input in_Xd_Y32_Z_Xd_Y2_Z [64];

with:
in.d[0].d[0] => in_Xd_Y32_Z_Xd_Y2_Z[0]
in.d[0].d[1] => in_Xd_Y32_Z_Xd_Y2_Z[1]
in.d[1].d[0] => in_Xd_Y32_Z_Xd_Y2_Z[2]
...
```
**Work around**

1. and 2. we currently fix by regex as a quick fix
for 3. i will see if i can fix it in the code otherwise one of my team it put on  regex+python and I fix all of it after tape out as a patch:)

rmanohar, 2022-04-29:
1-4 are quite straightforward; do you want me to take care of it?  For 1-2, I would use -m for ("mangle output"). It's too bad that the Verilog syntax for quoting names isn't supported.

For 3, we could add an option to generate this version v/s the current output format. -f for fuse signal flags?

For 4, we can add a config option that specifies a string prefix for globals (which is currently "top."), so that it can be set to whatever is needed for a specific simulator. How about verilog_global_prefix?

olerichter, 2022-05-02:
If you have time for it that would be obviously amazing, otherwise I will do it after we send the chip off 1.5 weeks and use it to learn more about the code base (we have working regex workarounds so its not urged, our system sims are running fine).
for 3 we actually realized an additional problem that the AMS runner has (we have a running workaround for that too):
if you have
```
output signal1;
reg signal1;
CELL instance_cell (.y(signal1));
```
the register directive needs to be removed, because it does not accept registers in wire like situations where they are constantly driven.
```
output signal1;
CELL instance_cell (.y(signal1));
```
for our work around we replace all `reg` that are not outputs with `wire`, and for output we just remove the line. 

but i am not sure if that has any regression problems, or defining it as reg is actually obsolete and other simulators are just ok with driving a reg constantly

rmanohar, 2022-05-04:
This is interesting... I believe I generate a reg for signals that are driven. I can make this another command-line option, and make it the default if we don't find any issues over time.

rmanohar, 2022-05-08:
Just added -f (fuse directives) and -m (mangle names) options.  Also, you can now specify an ACT config file option (act.global_signal_prefix) to change the global signal behavior. The default value is "top."

closed 2023-03-02 by rmanohar