asyncvlsi/act #47: ACT does not accept user-defined type from wiki

fhuemer, 2024-07-04:
**To Reproduce**
Steps to reproduce the behavior:
1.  Use the ACT source included below.
2. Run e.g. `act:read` in `interact` on the source file shown below
3. The following error message is produced:

```
interact> act:read mytype.act
ERROR: File `mytype.act', line 5, col 9
	Assignments only permitted to `self' or `selfack' in HSE body
```

**ACT source**
I have a single source file containing just the user type type definition from the [wiki](https://avlsi.csl.yale.edu/act/doku.php?id=language:types2) 

```
deftype mystruct <: int<16> (int<4> f1, f2; int<8> f3)
{
  methods {
   set {
     f1:=self >> 12;
     f2:=(self >> 8) & 0xf;
     f3:=self & 0xff
   }
   get {
     self:=(f1 << 12) | (f2 << 8) | f3
   }
  }
}
```

**Computer setup (please complete the following information):**
 - Fedora Linux, `act`  version `4d94dec` (I compiled the version in actflow)

rmanohar, 2024-07-04:
Thanks for catching this! This is a bug in the documentation (I just updated it). An earlier version of the language supported this particular feature, but it has since been removed from the language. (We found that it was quite confusing.)

fhuemer, 2024-07-05:
Thank you for clarifying this!

closed 2024-07-05 by fhuemer